wokwi/wokwi-docs #15: dark mode almost hides the switch component head

Puneethkumarsn, 2021-04-28:
[image]

can we have something like a border of white/gray around the black head?
So that, it will look good on both normal and dark mode screens?

urish, 2021-04-28:
Relevant file: [parts/wokwi-slide-switch-diagram.svg](https://github.com/wokwi/wokwi-docs/blob/master/docs/parts/wokwi-slide-switch-diagram.svg)

arcostasi, 2021-04-28:
> can we have something like a border of white/gray around the black head?
> So that, it will look good on both normal and dark mode screens?

Maybe some parameter to change the background color of the black head? red would look cool in dark mode

urish, 2021-04-28:
Yes mean in general? not just in the docmentation?

arcostasi, 2021-04-28:
> Yes mean in general? not just in the docmentation?
I can clearly see the black on the dark gray background of the wokwi simulator in dark mode, in the documentation that is a little harder to see

Puneethkumarsn, 2021-04-28:
> > Yes mean in general? not just in the docmentation?
> 
> I can clearly see the black on the dark gray background of the wokwi simulator in dark mode, in the documentation that is a little harder to see: https://wokwi.com/arduino/projects/288734316479906312?dark=1

Yes. Same opinion here.