wokwi/wokwi-docs #55: Serial.println stops working when accessing a structure in a structure?

Sarah-C, 2021-08-11:
https://wokwi.com/arduino/projects/306622481699963457

During setup(), a call's made to addFadeToQueue. On line 122, just after:        
    fadeLEDs[led].fadeQueue[head].startRGB.r = d.startRGB.r;
    Serial.println("This isn't displayed");
    return; // early return

The Serial.println command stops outputting anything. I have no idea why.

If I  put it BEFORE line 122, it displays fine.                  

    Serial.println("This isn't displayed");
    return; // early return
    fadeLEDs[led].fadeQueue[head].startRGB.r = d.startRGB.r;

It really bodges my attempts at super basic debugging practices. (Still learning GDB)

urish, 2021-08-11:
Hi Sarah! This usually means that something in "fadeLEDs[led].fadeQueue[head].startRGB.r = d.startRGB.r;" doesn't work quite right.

GDB would probably be the quickest way to debug it, but it has a steep learning curve.

Another suggestion would be try to and print the values of each of the variables: `led`, `head`, `fadeLEDs[led].fadeQueue[head]`, etc., in attempt to find out which one doesn't match with your assumptions / expecations.

Also, when adding a debug print, make sure to call `Serial.flush();` right after it - the Serial data is stored in a buffer and transmitted asynchronously, so this makes sure that all the characters in the buffer has been printed before moving on to the next command.

We have a pretty active community on Discord at https://wokwi.com/discord, so you may want to join as well. The members tend to be very helpful with this kind of questions.

sutaburosu, 2021-08-11:
@Sarah-C I haven't seen you on Discord, so I'll leave a brief note here.

Compiling your sketch gives the message `Global variables use 2074 bytes of dynamic memory.` The Uno has only 2048 bytes of RAM. Each FadeLED is 175 bytes. Your sketch runs fine after reducing NUM_LEDS to 5.