wokwi/wokwi-docs #57: LED Stripes missing?

ModischFabrications, 2021-08-13:
https://docs.wokwi.com/diagram-format doesn't refer to the different LED stripes available. 

wokwi-neopixel-canvas works, copied from an example, are others supported as well?

urish, 2021-08-13:
That's a good point. The reason I haven't written up the docs yet is that the names are a little bit messy, and I wanted to clean it up a bit before writing the docs.

Right now we have:

- wokwi-neopixel - a single pixel
- wokwi-led-ring - a ring with configurable size. [example](https://wokwi.com/arduino/projects/287302452979433992)
- wokwi-neopixel-matrix - a matrix with neopixel graphics
- wokwi-neopixel-canvas - similar to matrix, but different graphics which is easier on the CPU
- wokwi-neopixel-strip - doesn't really draw anything on screen, but used to define your own layouts using a custom iframe. Examples:
  - [Custom matrix by @arcostasi](https://wokwi.com/arduino/projects/289599145354723853)
  - [2D spirals](https://wokwi.com/arduino/projects/284916546575270412)
  - [3D spirals](https://wokwi.com/arduino/projects/284895354051101197)
  - [3D cube](https://wokwi.com/arduino/projects/278306947843752461)
- wokwi-neopixel-meter - doesn't really draw anything, just shows you the FPS / current consumption
- wokwi-splendida - custom board with spiral layout (https://wokwi.com/splendida for examples)

ModischFabrications, 2021-08-14:
Thanks for the extensive list, I can work with that :) 

Are there plans to map other strips as well? My code actually uses a SK9822, at the moment I'm manually patching the config to the supported stripe (WS2811?). It's not necessary, just asking for convenience sake.

urish, 2021-08-14:
APA102 is probably coming soon, I've just started a thread on the [discord server](https://wokwi.com/discord) to come up with a more consistent naming scheme for the smart LED parts. It might be something along the lines to `wokwi-<type>-<geometry>`, e.g. `wokwi-ws2812-ring`, `wokwi-apa102-matrix`, but we're still discussing and hopefully somebody comes up with some alternative idea so we can have something to compare against.

Specifically for the SK9822, are they using the same protocol as APA102? Or would that require a different implementation?

ModischFabrications, 2021-08-14:
The new naming scheme seems intuitive to me, I don't see any immediate problems with it. A more general hindrance for me is the "wokwi" prefix everywhere. A prefix for all might as well be for none, but I'm sure there is a reason for it. 

FastLED has a specific Chipset for the SK, which works for me (https://github.com/ModischFabrications/turn_indicator_code/blob/f89fb6c4f026df3b65143919f2e83ba02c0c54aa/src/lights.h#L186), but it think they are more or less equal. If I recall correctly choosing SK is backwards compatible to APA, but you might want to check with them directly.