wokwi/wokwi-docs #82: LEDs in serial wont work

nickelswitte, 2022-01-11:
Hello,

I am trying to make a blink example of two LEDs wired in serial, but it does not work.
[image]

It makes no difference, if I insert a resistor or not.

Is there something I am missing, or does this not yet work?

Thanks for the amazing program, highly appreciate the work!
Nick

urish, 2022-01-11:
Hi Nick, thanks for reporting and for the feedback!

Yes, this does not yet work. The simulation core is digital, so it does not know how to handle this type of circuit. 

What is your use case?

nickelswitte, 2022-01-11:
Thanks for the quick reaction!

I am actually developing a course for teaching uCs to kids. 
Our project is to create a small electronic dice that they can use to 'roll' a dice and see the results on LEDs.

You can imagine it (simplified of cause) like this:

[image]

The original circuit is this:
[image]

It would have been nice to do serial to have it exactly like the schematic, but it works like shown above as well.

Your website is actually a blessing. Teaching uCs to kids without a simulation wouldn't be possible at all.

Cheers