wokwi/wokwi-docs #95: Scrolling issue

arcostasi, 2022-02-15:
I was having this problem with scrolling in the Wokwi docs.

**Solution:** I turned disabled the option **Experimental Web Platform features** 
(_chrome://flags/#enable-experimental-web-platform-features)_)

**More details**: [Scrolling issue since beta.15](https://github.com/facebook/docusaurus/issues/6587)

urish, 2022-02-15:
Thanks for reporting Anderson!

In the meanwhile, I reverted to beta.14. Can you please check if this still happens?

arcostasi, 2022-02-15:
> In the meanwhile, I reverted to beta.14. Can you please check if this still happens?

Hi Uri, I enabled again the option **Experimental Web Platform features** and the problem did not occur! 
Scrolling works correctly again! thanks

urish, 2022-02-15:
Lovely, thanks for checking :)

closed 2022-02-15 by urish