wokwi/wokwi-docs #120: How can I translate wokwi-docs to Chinese?

breakeryu, 2022-07-03:
Hi!
I want to translate this doc to Chinese. How can I do this?

breakeryu, 2022-07-03:
I think that I can join this project and share this great website with my friends.

urish, 2022-07-07:
Thank you! What is your background? How familiar are you with GitHub?

@zhufree started to work on Chinese translation a few months ago, and translated a few pages (e.g. the [diagram editor page](https://github.com/wokwi/wokwi-docs/blob/master/i18n/zh-CN/docusaurus-plugin-content-docs/current/guides/diagram-editor.md)), but there's still much work to do - update the existing translations (they're more than 6 months old), and translate additional pages.

How would you like to work on it?

zhufree, 2022-07-09:
Yes you can contact me for more information

breakeryu, 2022-07-15:
> Yes you can contact me for more information

hi! zhu, I have sent an e-mail to your Gmail box.

closed 2022-07-22 by breakeryu