wokwi/wokwi-docs #122: adding explicit examples that show the use of the attributes in EACH example-code

StefanL38, 2022-07-13:
Hi,
the headline says it all
adding explicit examples that show the use of the attributes in EACH example-code

The example for the PIR-Sensors has empty attributes 
`{
      "type": "wokwi-pir-motion-sensor",
      "id": "pir1",
      "top": -42,
      "left": 25.81,
      "attrs": {}
    },`

that is a somehow bad example as it dos NOT explain how to use the attributes
you should add an example that shows all attributes
or at least a link where the pattern how to use attributes is explained

urish, 2022-07-20:
Thanks for the feedback! For starters, we can add an explainer in [diagram format](https://docs.wokwi.com/diagram-format), and link to it from the Attributes section of each part.

letsgo88, 2023-01-25:
7hours later i find your comment and change the anode to a cathode and now everything works. [StefanL38](https://github.com/StefanL38) i could kiss ya right now you made my day. I knew what the problem was just had no idea how to edit it, until i read your comment. This is defiantly needed so people don't give up before they find the potential of the site. 
and Wokwi if you give your instruction to someone that has never used the site before they will quickly show you if your instruction are good or far to vague.

matititam, 2023-01-26:
Hello @letsgo88

I'm glad you were able to find a solution, but it's unfortunate that it took you 7 hours 😓. Can you help us understand how we can improve our documentation and processes to prevent this from happening for new users in the future?

Can you please tell me where(site, forum, particular wokwi guide, or anything) did you try to find the solution to start with? What all options did you explore? 

Thanks in advance 😃

letsgo88, 2023-01-26:
Thankyou you for your interest I will be happy to answer your questions . But with my current deadline looming I will need to wait untill my schedules free up, before I help improve the service you provide. Please be patient for my reply.Many thanks Chris