wokwi/wokwi-docs #163: VS Code / Redundant path in `wokwi.toml`

Semnodime, 2023-02-07:
wokwi refuses to simulate the project when one of `firmware` vs `elf` is omitted while they are identical.
I propose the option to merge them / ignore the missing one by assuming it as being redundant.

https://github.com/wokwi/wokwi-docs/blob/cf32894c6f3be943b8bea1d72904ba3eaa8f3270/i18n/pt-BR/docusaurus-plugin-content-docs/current/vscode/project-config.md?plain=1#L18-L19

----
This only happens when deleting the whole line within the .toml file….

urish, 2023-02-08:
Good suggestion! May make sense to remove `elf` when firmware and elf are they same.

thubalek, 2025-06-16:
I'm totally confused with documentation. What is the meaning of those two entries? I didn't find any explanation for them. And Visual Studio extension complains that both options have to be strings.

urish, 2025-07-06:
Released version 2.7.0 of the VS Code extension, and the `elf` field is now optional as suggested.

closed 2025-07-06 by urish