wokwi/wokwi-docs #167: SevenSegmentTM1637 library and the Docs.

Koepel, 2023-02-26:
(1)
The SevenSegmentTM1637 library has a problem. A bug was fixed, but the library version was not updated. Therefor the Arduino IDE and Wokwi take in the version with the bug. Some examples run in Wokwi, but if an other example is tried then it does no longer work.
New users could be confused.

The library: https://github.com/bremme/arduino-tm1637
The bug is the word "bouchingBall" (a missing 'n') in the file SevenSegmentFun.h

Should the example with the library be removed from the Docs ? https://docs.wokwi.com/parts/wokwi-tm1637-7segment

(2)
In the Docs is the text: "_Here are some TM1637 libraries you can use on Arduino: SevenSegmentTM1637@1.0.0, Grove 4-Digit Display_". 
I don't know what that means.

urish, 2023-02-26:
> SevenSegmentTM1637@1.0.0

"SevenSegmentTM1637@1.0.0" means version 1.0.0 of the library, which does work well. 

This is also the version we're using in the docs.

Ideally, if someone republished the latest version of the library under a different name to the Arduino Library Index, that would both solve the issue and allow users to use the latest version (and we could update the examples).

We could also just remove the example / reference from the docs

Koepel, 2023-02-26:
If someone has the right version in Wokwi and then makes the real project in the Arduino IDE, then it no longer works. That is not straightforward for new users.
You could remove the example or explain it in the Docs.

FernandoGarcia, 2023-02-27:
Hi!

I think @urish could allow use of git path in library.txt as PIO does in lib_deps then use git clone.

I could point to a fork with some bug fix to an abandoned repository.

Change tutorials or examples for bugs in abandoned repositories will be more problematic.

Best regards.

Koepel, 2023-02-28:
Could an example with the RT1637_RT library be added ? That library is actively maintained.
This is my example: https://wokwi.com/projects/356661328560439297
To make the example as a simple introduction, the clock format was added for it: https://github.com/RobTillaart/TM1637_RT/issues/24

urish, 2023-02-28:
Yes, this is great! I replaced SevenSegmentTM1637 with RT1637_RT, and added your example instead of the SevenSegmentTM1637 example.

closed 2023-02-28 by urish

FernandoGarcia, 2023-02-28:
In Brazil we say: It's like cove the sun with a sieve.

You will do the same with all libraries?

[image]

What you will do with libraries available in library manager but no present in Wowki docs?

urish, 2023-02-28:
Right! I think it makes sense to make sure all examples in the docs are functional and are easy for users to build upon.

But are we definitely not aiming to fix the entire Arduino library index ecosystem (which has 5583 libraries at the moment!)

FernandoGarcia, 2023-02-28:
Maybe my suggestion is against your business model but it could be an alternative to solve almost all problems with deprecated libraries in library manager.

urish, 2023-02-28:
Fernando, are you referring to this suggestion?

> I think @urish could allow use of git path in library.txt as PIO does in lib_deps then use git clone.

This is something that I've considered in the past, but the issue with git clone is that it's relatively slow. The build server slot is limited to 60 seconds per user, and we're already seeing about 0.4% of the builds failing due to time out.

GitHub also makes it possible to host huge repositories with multiple gigs, and they may cause the server to run out of disk space. 

All these problems have solutions, of course, but it'll require a non trivial amount of engineering to solve them. For more complex builds, it makes more sense to use something like PIO, and then Wokwi for VS Code comes into play.

I hope this explanation helps!

FernandoGarcia, 2023-02-28:
Yes, thanks for your answer.

I can understand now. 

I think PIO a little bit complicated to noobs but it's an alternative.

I have used the extension in VSCode it's works better than on browser.

When the extension allow circuit edition it will be very good.

maxint-rd, 2025-03-03:
Hmmm... as of today it seems that the wokwi-tm1637-7segment part is broken. Yesterday it was still okay; today none of the projects or examples work anymore with any library on none of the platforms. The simulator start with two segments of digit 0 lit. After init all segments light up, but nothing else happens.

Am I the only one? Is it my slow internet connection? Any clue about what happened?