wokwi/wokwi-docs #186: Adding Files to a Project

brightproject, 2023-08-28:
I can't recover my Discord password - that's why I'm writing about problems here.
1. I simulate my project that runs on real hardware.
Added 16 "*h" files and can't save project.
[image]
Is this a limitation of the free version or a bug?
I managed to save the project archive - help me figure out why it is not saved and why the buttons do not always work adequately.
The project is quite interesting, and can greatly popularize your simulator.
Please see it if possible.
Any questions are welcome.
[Flightinstruments_TFT_eSPI.zip](https://github.com/GyverLibs/EncButton/files/12448649/Flightinstruments_TFT_eSPI.zip)
2. Strange work with the button library *, the code compiles every other time.
Issues with internal pullups, does the simulator not simulate internal pullups well?
3. The display * displays the image shifted to the right, you can see it in the screenshot above.

urish, 2023-08-29:
> Cant add a EncButton library with version newer than 2
> [image: image]
> <https://user-images.githubusercontent.com/84599917/264039168-e72b0447-8aec-4f9c-a9c9-62013c2faa5b.png>
> [image: image]
> <https://user-images.githubusercontent.com/84599917/264039035-e4ee73ab-d87b-492b-8b94-c585be926d85.png>
> [image: image]
> <https://user-images.githubusercontent.com/84599917/264039403-531b6100-010b-4623-8a88-91ff6d170190.png>

Thanks for reporting, will look into it

urish, 2023-08-29:
@GyverLibs should be fixed now, can you please try again?

urish, 2023-08-29:
Thanks for the feedback @brightproject!

1. I haven't had a chance to look at the project yet, but saving the project may fail if the total size of source code is greater than 1MB (only saved source code - not counting external libraries). This may happen if the .h file include large images.
3. We updated the libraries on the build server. Does it work more reliably now?
4. The image is not shifted to the right - the physical board has a black margin next to the display, so the simulation image also includes this margin. So this is by design. 
5. I'm curious - what issues are you having with the VS Code version?