wokwi/wokwi-docs #201: [Documentation] Example to show static and multiplexed signals for 7-segment displays.

Koepel, 2024-05-11:
Documentation: https://docs.wokwi.com/parts/wokwi-7segment
My example: https://wokwi.com/projects/397598198000164865

Could my example be added to the documentation ?
It is to show the 7-segment displays controlled by either static signals or multiplexed signals without libraries to show the difference.

closed 2024-05-12 by urish

Koepel, 2024-05-12:
It is called "7-Segment without a library on Arduino Mega", but that is not its purpose.
In the documentation, it is mentioned that a single digit requires 8 GPIO pins and multiple digits can be tricky. My example is to show the difference between non-multiplexed signals with static digital GPIO pins and multiplexed signals which need to keep the display updated with a delay.

Could you change the text ?

urish, 2024-05-12:
Can you please suggest an alternative text?

Koepel, 2024-05-12:
I gave the Wokwi project the title: Static versus Multiplexed signals for 7-segment displays.

commit 8895710b (2024-05-12): docs(wokwi-7segment): add no library example (static/mux) on mega #201

close #201