wokwi/wokwi-docs #225: Broken license email from your website

VorpalBlade, 2025-04-01:
Hi, I did not find anywhere else to report bugs except on github. Please add a general contact functionality and forward this to the proper person to deal with this as I'm sure the docs repo is not the proper location.

I tried using the vscode extension (for some personal / open source Rust code) and found that I had to activate it, I selected to activate with an email and generate a one-use email alias on my self hosted email server. Unfortunately the email you sent contains no link, at least not in the text version of the email, and as I happen to use a terminal based email client I cannot see the HTML email content.

Please fix your emails so that you include a link in the plain text body of the multipart message.

urish, 2025-04-01:
Hi, thanks for reporting. The links are generated by a third party service from Google, called Firebase Authentication.

Can you please paste the content of the text version of the email?

VorpalBlade, 2025-04-01:
Certainly:

```
Hello,

We received a request to sign in to Wokwi using this email address, at 2025 April 01 18:08 UTC. If you want to sign in with your email@redacted.no.spam.please account, click this link:

Sign in to Wokwi

If you did not request this link, you can safely ignore this email.

Thanks,

Your Wokwi team
```

Notably "Sign in to Wokwi" was just plain text, not a link. I don't know what you need to do to convince the third party service to include an actual URL in the message body. Perhaps this is why so many emails of this type has something like "if you cannot click the link try: https://example.com/very_long_url?with=lots&of=parameters" towards the end of the message?